DiamondLightSource/dodal #6: Find a good way to keep beamline and simulator devices in step

DominicOram, 2022-08-31:
As a developer/scientist I would like to make to make sure that the devices for the simulated beamline are kept the same as those for the real one, with as much ease as possible. Currently to add a new device to e.g. i03/s03 I need to add the following to `i03.py`:
```python
 BEAMLINE_PREFIX = "BL03I"

def device():
     return MyDevice(prefix=f"{BEAMLINE_PREFIX}-BLAH")
```

Then the following to `s03.py`:
```python
 BEAMLINE_PREFIX = "BL03S"

def device():
     return MyDevice(prefix=f"{BEAMLINE_PREFIX}-BLAH")
```

Which is an exact replica, except the `BEAMLINE_PREFIX`. Ideally I should just need to add to `i03.py` and `s03.py` will update itself.

DominicOram, 2022-08-31:
Originally I had something like:
```python
from dodal import i03
i03.BEAMLINE_PREFIX = "BL03S"
i03.INSERTION_PREFIX = "SR03S"

from dodal.i03 import *
```
in `s03.py` but this causes big issues if importing `s03` and `i03` together. Alternatives could be to change the structure of `i03.py` into a class?

DominicOram, 2022-09-09:
We might also want a `fake_i03` that has all the i03 devices but as ophyd fake devices

DominicOram, 2023-02-14:
As discussed with @callumforrester we can just do this with an env variable. We can check `BEAMLINE` and use i03 if it is that, otherwise s03

stan-dot, 2024-09-25:
reading through this the best-case scenario to solving this is just adding a piece of documentation for this workflow. What do you think @DominicOram ?

similar topic to https://github.com/DiamondLightSource/dodal/pull/678

DominicOram, 2024-09-25:
Yes, I think probably covered by https://github.com/DiamondLightSource/dodal/issues/502 and https://github.com/DiamondLightSource/dodal/pull/678

closed 2024-09-25 by DominicOram