DiamondLightSource/dodal #11: Add usage statistics

DominicOram, 2022-09-05:
An issue we have in GDA is that it's not always clear what code is in use by what beamlines and so whether code can be removed or how changing it may affect other beamlines.

## Acceptance criteria
* By default people who use dodal get usage statistics of their use

## Note
Should probably use the same system as https://github.com/DiamondLightSource/python-dodal/issues/10

stan-dot, 2024-06-24:
open telemetry might provide this for free, which is now being added to [bluesky in this PR](https://github.com/bluesky/bluesky/pull/1724). 

Alternatively this might be provided by a [git server issue in blueapi](https://github.com/DiamondLightSource/blueapi/issues/511#issuecomment-2185877745).

The third alternative is to use a [code search engine, like numpy does](https://numpy.org/neps/nep-0023-backwards-compatibility.html#impact-assessment). 

From the looks of it the first option is the most automated one, therefore favoured for the lowest maintenance cost.

stan-dot, 2024-08-19:
@keithralphs what do you think?