DiamondLightSource/dodal #12: Think about how to store metadata

DominicOram, 2022-11-03:
For some devices there may be metadata that is required alongside the device for later storage. e.g. the material the device is made of. It probably makes sense for `dodal` to have this data for applications to access but the way it's stored etc. is TBD

## Acceptance Criteria
* A meeting is had to discuss the best way of storing this data
* The decision, and reasoning is documented in dodal with an example

stan-dot, 2024-07-09:
this might be related https://github.com/DiamondLightSource/blueapi/issues/537

Villtord, 2026-01-20:
Hmm, good call! I have no idea what Joseph meant exactly by "SoftSignals were temporarily added to some devices to allow for data to propagate to NeXus" and if they were removed already. There is no branch linked or examples given.
I guess if it a static attribute of a device which is unlikely going to change then having it as a soft signal is ok as long as it is added to describe or read signals and effectively will end up in a nexus file?

Villtord, 2026-01-21:
I personally think it is fine to leave them in soft signals as long as it is static metadata for a device that is unlikely going to change anytime soon (or at all).