DiamondLightSource/dodal #13: Baton management

DominicOram, 2022-11-03:
To avoid multiple people trying to write to the beamline at the same time we should have a baton. It probably makes sense for this baton to be in `dodal` so that anyone using it obeys the baton

## Acceptance Criteria
* There is a discussion about how to handle this in dodal
* There is some documentation on what the solution looks like

stan-dot, 2024-09-25:
this is not a problem in blueapi as far as I am aware. 

can we define the scenario in more concrete terms? Is it about different RE instances?

DominicOram, 2024-09-25:
The main thing we're trying to avoid is GDA and Hyperion trying to use the beamline at the same time. Therefore there needs to be an external baton that each application looks at to tell them if they're in charge or not.

stan-dot, 2024-09-25:
we could build a low level lock-and-baton system with redis, or in an etc-d key.

third party solutions include:
- https://developer.hashicorp.com/consul/docs/dynamic-app-config/sessions - (https://github.com/hashicorp/consul)
- https://zookeeper.apache.org/doc/r3.1.2/recipes.html (https://github.com/apache/zookeeper)

@tpoliaw  I searched on slack for messages with the `baton` keyword and thought you  may have suggestions - also amygdala might need to coordinate to make this work