DiamondLightSource/dodal #20: Reinstate filterwarnings in toml file

noemifrisina, 2023-02-16:
Reinstate the following lines turning warnings into errors in the `pyproject.toml` file. 
```
# https://iscinumpy.gitlab.io/post/bound-version-constraints/#watch-for-warnings
filterwarnings = "error"
```

It has been temporarily removed in #17 in order to proceed with tests.

stan-dot, 2024-09-25:
@noemifrisina  at the moment that field for `tool.pytest.ini_options` is such:

```toml
filterwarnings = [
    "error",
    # Ignore deprecation warning from ophyd_async
    "ignore:dep_util is Deprecated. Use functions from setuptools instead.:DeprecationWarning",
    # Ignore deprecation warning from zocalo
    "ignore:.*pkg_resources.*:DeprecationWarning",
    # Ignore deprecation warning from setuptools_dso (remove when https://github.com/epics-base/setuptools_dso/issues/36 is released)
    "ignore::DeprecationWarning:wheel",
]```

Should this be closed?

noemifrisina, 2024-09-25:
Yeah, I opened this issue right after setting up dodal, it looks like it's outdated now

stan-dot, 2024-09-25:
closing as already done

closed 2024-09-25 by stan-dot