DiamondLightSource/dodal #57: Add enums for detector motion

DominicOram, 2023-05-17:
A lot of the movements in `DetectorMotion` are binary setpoints, using 0/1 to mean open/closed. We should refactor these to be enums.

stan-dot, 2024-09-25:
```python

from ophyd_async.core import StandardReadable
from ophyd_async.epics.motor import Motor

class DetectorMotion(StandardReadable):
    """Physical motion for detector travel"""

    def __init__(self, prefix: str, name: str = "") -> None:
        self.y = Motor(prefix + "Y")  # Vertical
        self.z = Motor(prefix + "Z")  # Beam

        super().__init__(name)

```

at the moment the file does not have binary setpoints so either it's the wrong file or it has changed substantially since.

@noemifrisina from git log I infer you might have more context here, what happened here?

noemifrisina, 2024-09-25:
I'm not sure, I think that the original issue may have been referring to this [DetectorMotion device](https://github.com/DiamondLightSource/dodal/blob/main/src/dodal/devices/detector/detector_motion.py) ?
The i24 one is separated because it's quite different and is fine as it is at the moment...

stan-dot, 2024-09-25:
ah I see 

```python

   self.shutter_closed_lim = epics_signal_r(
            float, f"{prefix}{device_prefix}CLOSE_LIMIT"
        )  # on limit = 1, off = 0
        self.shutter_open_lim = epics_signal_r(
            float, f"{prefix}{device_prefix}OPEN_LIMIT"
        )  # on limit = 1, off = 0
        self.z_disabled = epics_signal_r(
            float, f"{prefix}{device_prefix}Z:DISABLED"
        )  # robot interlock, 0=ok to move, 1=blocked
        self.crate_power = epics_signal_r(
            float, f"{prefix}{pmac_prefix}CRATE2_HEALTHY"
        )  # returns 0 if no power
        self.in_robot_load_safe_position = epics_signal_r(
            int, f"{prefix}{pmac_prefix}GPIO_INP_BITS.B2"
        )  # returns 1 if safe
```