DiamondLightSource/dodal #74: systematise what to do with PV prefixes

dperl-dls, 2023-05-26:
- In general, PV prefixes belong where devices are instantiated, in the beamline files
- Sometimes this is not possible because multiple prefixes are needed (https://github.com/DiamondLightSource/dodal/issues/73)

dperl-dls, 2023-05-31:
Devices which take more than one prefix for different components should accept these as additional kwargs so that the root PV prefix should always be in the beamline files, for improved reusability of devices. E.g. use of `-EA-` and `-MO-` differs between beamlines and this should not prevent device reuse.

stan-dot, 2024-09-25:
applicable to the pressure cell - the one with separate ADC prefixes.

Might be of interest to: @barnettwilliam