DiamondLightSource/dodal #128: Beamline test seem very flaky

keithralphs, 2023-07-26:
I tried to add ad test for the p38 beamline definition using the make_all_device function as in test_i24.py, but was plagued by hangover of shared state from previous tests which was different from run to run on Github (due to the (correct) random order of test running) causing the tests to fail in different versions of Python on subsequent runs of the same actions. This seems to be associated with the BL and ACTIVE_DEVICES objects from the beamline_utilities module. Basically the set beamline can persist from one beamline test module to the next as can davices stored in the ACTIVE_DEVICES dict, resulting in tests that compare the length of created device to this dict failing randomly.

stan-dot, 2024-07-09:
idea - we use labels in dodal to indicate which beamline does an issue refer to. could be multiple and it's more reliable and easy to see and use than putting this sometimes in the title, sometimes in the description

stan-dot, 2024-07-09:
@keithralphs  how to replicate the issue?

stan-dot, 2024-07-10:
as discussed in person with @keithralphs , this is no longer relevant, as tests have been known to pass.

closed 2024-07-10 by stan-dot