DiamondLightSource/dodal #146: Should: There are times that we may want the whole edge array (for grid detection). We can do this in a new issue if you like but we will need it

Tom-Willemsen, 2023-08-18:
Should: There are times that we may want the whole edge array (for grid detection). We can do this in a new issue if you like but we will need it

_Originally posted by @DominicOram in https://github.com/DiamondLightSource/dodal/pull/143#discussion_r1297604042_
            
After some discussion in-person: needs a bit more thought about exactly what we want to expose (area over which to do a grid scan?)

DominicOram, 2023-08-18:
Currently we use the edge arrays to:
* Work out the area to do a grid over, done in https://github.com/DiamondLightSource/python-artemis/blob/acb726f2231481f221ad2af338c5961839e4372f/src/artemis/experiment_plans/oav_grid_detection_plan.py#L54
* Do an OAV centre, done in https://github.com/DiamondLightSource/dodal/blob/main/src/dodal/devices/oav/oav_calculations.py, but not actually used on a beamline currently

We need to get them somehow from the device but it's an open question whether the device should also do the calculations above

dperl-dls, 2024-04-03:
this was done in https://github.com/DiamondLightSource/dodal/pull/358

closed 2024-04-03 by dperl-dls