DiamondLightSource/dodal #149: Ophyd V2 devices: consider whether the current `device_instantiation` abstraction(s) are correct

Tom-Willemsen, 2023-08-18:
For example, V2 devices don't necessarily take a prefix but have the prefix as part of the name. 

We could:
- Say that all of *our* v2 devices take a prefix regardless, keeping consistency with v1
- Refactor `device_instantiation` to still be semi-generic but build name/prefix combinations correctly for v1 or v2 as appropriate
- Split `device_instantiation` into `_v1` and `_v2` with different behaviour.

Unclear to me exactly what is desirable.

stan-dot, 2024-08-19:
@coretl this is relevant to #483 and can be either incorporated, closed, or still kept as-is

coretl, 2024-08-20:
I think this can be closed as not relevant after #483 

@DominicOram do you agree?

DominicOram, 2024-08-20:
Yh, happy to close. Tbh we won't have any v1 devices round soon anyway

closed 2024-08-20 by DominicOram