DiamondLightSource/dodal #188: Zebra failing to put complete

DominicOram, 2023-10-05:
We're still getting some timeout issues on the put complete of the zebra despite increasing the timeout in https://github.com/DiamondLightSource/dodal/issues/171.

DominicOram, 2023-10-05:
Plan is to try and catch this in the debug logs but we will need some form of https://github.com/DiamondLightSource/hyperion/issues/910 for that

stan-dot, 2024-09-25:
@DominicOram , how to replicate the issue? the timeout is still at 3.0 as it was in the #171 

hyperion 910 is merged by now

DominicOram, 2024-09-25:
We have no idea how to replicate the issue. It happens sporadically in the middle of the night. We added a retry in the hyperion side (https://github.com/DiamondLightSource/hyperion/pull/1124) and we do not seem to have seen it since then. I would still like to get to the bottom of it but it's low prioirity