DiamondLightSource/dodal #222: Add support for 3.12

DominicOram, 2023-10-29:
Python 3.12 is now out, we should add support for it to keep up to date.

Depends on https://github.com/DiamondLightSource/dodal/actions/runs/6642933356/job/18074448752?pr=216

This will also be hampered by us not yet having python/3.12 in the module load system.

## Acceptance Criteria
* Non-test builds are unpinned (e.g. reverse https://github.com/DiamondLightSource/dodal/pull/221/commits/b980f9bf3df0e0042a650dd9d46540d12f7589ab)
* CI runs tests under 3.12 
* CI all still passes

stan-dot, 2024-07-09:
update:
original p4p PR was closed.

This PR is where it seems the cutting edge is
https://github.com/mdavidsaver/p4p/pull/141

stan-dot, 2024-08-19:
latest `main` branch seems to support 3.12 https://github.com/DiamondLightSource/dodal/blob/main/pyproject.toml

can we close this @DominicOram ?

DominicOram, 2024-08-19:
Not until CI is in properly https://github.com/DiamondLightSource/dodal/blob/46132f0e066debaf906c1a80dcefca175b3f5c01/.github/workflows/ci.yml#L24

stan-dot, 2024-08-19:
ah right I forgot about this

PR #859 (2024-10-23): Add python 3.12 to CI
Fixes #222

### Instructions to reviewer on how to test:
1. Verify tests pass
closed 2024-10-23 by dperl-dls