DiamondLightSource/dodal #233: Split tests into system/unit test folders

dperl-dls, 2023-11-13:
As with Hyperion https://github.com/DiamondLightSource/hyperion/issues/970 it would be nice to be able to run all the unit tests at once from vscode. For this, we need to separate the system (`@pytest.mark.s03`) and unit tests at the top level of the tests directory

## Acceptance Criteria
* Directory structure looks like:
```
tests/
  unit tests/
    devices/
    beamlines/
    ...
  system tests/
    devices/
    beamlines/
    ...
```
* you can run all the unit tests with one click in vscode

stan-dot, 2024-09-25:
alternatively just adding the `@pytest.mark.system` might do the trick without a directory rearrangement

dperl-dls, 2024-10-07:
We used to use that solution in hyperion but it was annoying not to be able to run only the unit tests from the test explorer in vscode, which is why we switched

stan-dot, 2024-10-07:
testing now in Dodal

in the test explorer with running `all` I get 655/678 on the main branch and yeah, adding arguments into the filter of sorts `!s03` does not seem to impact the filter.

dperl-dls, 2024-10-07:
yes, I wish the test runner had a way to add pytest options or marks. if you want to make an improvement to vscode-python or whatever that's in I'd consider it an alternative solution for this issue ;)

PR #822 (2024-10-08): Move system tests to a separate directory
Fixes #233

### Instructions to reviewer on how to test:
1. Open the vscode test discovery tool
2. one-click run the tests that are CI-compatible
3. one-click run the s03 tests (which now are in a separate folder)

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://diamondlightsource.github.io/dodal/main/reference/device-standards.html)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this dev [... cut]
closed 2024-10-08 by stan-dot