DiamondLightSource/dodal #247: No dev option for `get_beamline_based_on_environment_variable()` in `dodal.utils.py`

olliesilvester, 2023-11-29:
This function looks for the BEAMLINE environment name and then tries to import the associated dodal module. When running in dev mode, the BEAMLINE environment variable isn't set, and there would be no module to import. This means we are currently unable to start hyperion in dev mode. A quick solution would be to default to importing the i03 module if we're running in dev mode.

## Acceptance Criteria
* Implement a way to get `get_beamline_based_on_environment_variable() ` in developer mode
* Ensure `hyperion.sh --skip-startup-connection` runs succesfully after doing this change (with no errors in` hyperion/tmp/dev/start_log.txt`)

stan-dot, 2024-09-25:
also relevant for ViSR aka BL01C aka b01-1

olliesilvester, 2024-10-28:
Sorry, I only just saw this for some reason. This ticket was originally for Hyperion's dev mode, but we never got around to fixing it as running Hyperion locally wasn't massively useful anyway

stan-dot, 2024-10-28:
so is it no longer required?

olliesilvester, 2024-10-28:
It might still be useful at some point to get hyperion/mx-bluesky working locally, just not very urgent