DiamondLightSource/dodal #252: Revisit use of 'call_in_bluesky_event_loop'

dperl-dls, 2023-12-01:
To instantiate `ophyd_async` devices and to do any necessary synchronous actions, we are using  'bluesky.run_engine.call_in_bluesky_event_loop' which will fail if there is no active `RunEngine`. This is an issue for testing, as we can't instantiate devices on their own, and means weird stuff like having to have test fixtures in the right order so that an `RE` exists before devices are created.

## Acceptance Criteria
* There exists some useful utility function which will use `bluesky.run_engine.call_in_bluesky_event_loop' if available, but not fail if it isn't

DominicOram, 2024-03-18:
Related to https://github.com/DiamondLightSource/hyperion/issues/1263?

dperl-dls, 2024-03-18:
yes, when we know what the long-term solution for this will be in ophyd-async we can resolve both of these the same way