DiamondLightSource/dodal #253: Make a structured array for zocalo results and use the correct descriptor

dperl-dls, 2023-12-01:
After https://github.com/DiamondLightSource/hyperion/pull/1013 and https://github.com/DiamondLightSource/dodal/pull/250 we will not have descriptor documents which specify the correct datatype for the zocalo results. We should follow up on https://github.com/bluesky/event-model/issues/214 and use the solution there to fix this

## Acceptance Criteria
* zocalo results have a descriptor which correctly matches their datatype

DominicOram, 2024-05-07:
Also for aperture scatterguard

stan-dot, 2024-12-02:
both 'base' PRs have been since merge, the [linked event model issue](https://github.com/bluesky/event-model/issues/214) hasn't seen activity since December 2023