DiamondLightSource/dodal #257: Dodal (and hyperion) loggers should have TRACE level logging facility

rtuck99, 2023-12-07:
logging for dodal (and hyperion, but dodal especially) should have the facility to log at TRACE level.
Since INFO is the default log level for production to get sensible log output for high-level functionality, all high-volume logging messages are output at DEBUG as it is the only currently existing more detailed log level. This leads to excessive log-spam when debug logging is enabled and we don't have a way to get debug logging without wading through lots of low-level ophyd/dodal reads and writes.

We should also have independent control of the log-level for bluesky and ophyd from dodal and hyperion, so that we don't have to see their log messages as well, via a standard logging config file.

DominicOram, 2023-12-08:
I don't disagree with this issue but it also feeds into a wider discussion on how we want to do logging. e.g. see https://github.com/DiamondLightSource/hyperion/issues/910

dperl-dls, 2024-02-06:
After looking into this for https://github.com/DiamondLightSource/hyperion/issues/910, I think it will be too messy to add this, but the required functionality should be there after the logging changes there

closed 2024-02-06 by dperl-dls