DiamondLightSource/dodal #261: Skip pin tip device on i03 system test

DominicOram, 2023-12-11:
Currently running `python tests/system_tests/test_i03_system.py` does not work as the i03 pin tip detect device cannot connect. This is because:
* It is an ophyd device and we're not set up to connect to them on system tests
* It is a PVA device and on the beamline without a PVA gateway

## Acceptance Criteria
* There is a generic way to skip some devices in the system tests

DominicOram, 2024-04-24:
@coretl - am I right in saying there's no plan for PVA gateway being rolled out?

coretl, 2024-04-29:
> @coretl - am I right in saying there's no plan for PVA gateway being rolled out?

That is correct, it is on the list of things to do for D-II, but there is no date on this yet. Definitely not this year.