DiamondLightSource/dodal #307: Mako

callumforrester, 2024-01-25:
There should be an ophyd-async implementation of the mako detector in dodal. Useful for I22 and the test rigs.

## Acceptance Criteria
* Mako detector class written
* Capable of configuring a mako for hardware or software triggered scans

Relm-Arrowny, 2024-11-29:
We have multiple mako g-125 on i10, I could have a crack at this, if this still needed?

callumforrester, 2024-12-02:
We now have an [AravisDetector](https://github.com/bluesky/ophyd-async/blob/main/src/ophyd_async/epics/adaravis/_aravis.py) class in ophyd-async that should cover makos, however due to the variety of camera versions and IOC versions we often find ourselves having to update it to cope with a new version.

Try it out on the i10 cameras and see if it works, see [example usage](https://github.com/DiamondLightSource/dodal/blob/b4c87ae02218db2ce11e906a4b25066a3597476a/src/dodal/beamlines/training_rig.py#L48).

Relm-Arrowny, 2024-12-02:
> Try it out on the i10 cameras and see if it works, see [example usage](https://github.com/DiamondLightSource/dodal/blob/b4c87ae02218db2ce11e906a4b25066a3597476a/src/dodal/beamlines/training_rig.py#L48).

I do not really need it until I want to auto align the diffractometer pin which is something gda current does not do atm, but just for record keeping, [AravisDetector](https://github.com/bluesky/ophyd-async/blob/main/src/ophyd_async/epics/adaravis/_aravis.py) does not current work on mako g-125 at i10 due to mismatch epicPv and enum. 
[image]

callumforrester, 2024-12-02:
Hmm, that might require an IOC update, I suggest sticking on the relevant controls engineer's backlog