DiamondLightSource/dodal #308: Deprecate plans/stubs that decorate Bluesky built-in plans/stubs with type hints, once type hints on upstream functions

DiamondJoseph, 2024-01-26:
After #304 adds wrapped versions of Bluesky plans and stubs (to enforce type annotation requirements), and after [bluesky#1610](https://github.com/bluesky/bluesky/pull/1610) or [bluesky#1600](https://github.com/bluesky/bluesky/pull/1600) adds type annotations to the upstream functions, deprecate the wrapped functions and replace calls to them with calls to the upstream functions.

stan-dot, 2024-12-02:
the first connected PR has been completed, the other two have had their goals completed with this PR: https://github.com/bluesky/bluesky/pull/1712

arguably this could be work on now. @DiamondJoseph what do you think?

DiamondJoseph, 2024-12-03:
The plan stubs in the bluesky.plans module are overly tight when ophyd-async devices are mixed with ophyd devices, so it's not quite doable yet. We also still need to limit which plan stubs are exported as some don't make sense to run individually (e.g. opening a run may leave the beamline in a bad state to run the next plan without any feedback).

DominicOram, 2025-05-22:
Where are we at with this? If the blocker to removing the wrapped stubs is just `ophyd`/`ophd-async` mixing the only `ophyd` device we have that I'm aware of is the `Eiger` and this is only used by MX, who do not currently use these stubs. I would be happy for them to be removed, particularly as they are causing us maintenance overhead (https://github.com/DiamondLightSource/dodal/pull/1240)

DiamondJoseph, 2025-05-22:
We could potentially move them into blueapi, but the versions that are wrapped still use variadic args that we can't serialize/deserialize or are too broad:

```python
@plan
def abs_set(
    obj: Movable,
    *args: Any,  # not deserializable
    group: Optional[Hashable] = None,  # not deserializable 
    wait: bool = False,
    **kwargs,  # not deserializable
) -> MsgGenerator[Status]:
```

DominicOram, 2025-05-22:
Is there a `bluesky` issue to make into something you can use or is that not going to happen?

DiamondJoseph, 2025-05-22:
The kwargs eventually get passed through to the Device set method, and there could be any number of "Movable in name only" Ophyd devices in use that aren't compliant with the protocol at other facilities. I think type hinting the args which is then ignored at facilities not making use of type checking is unfortunately probably as good as we're ever going to get.