DiamondLightSource/dodal #340: Improve MakoDriver deadtime handling.

DiamondJoseph, 2024-02-16:
Mako and Manta driver usually used with G-234 and G-235 cameras respectively, but exceptions exist.

Deadtime for [the supported models of cameras ](https://cdn.alliedvision.com/fileadmin/content/documents/products/cameras/various/appnote/GigE/GigE-Cameras_AppNote_PIV-Min-Time-Between-Exposures.pdf) should be extracted to a config file and read only as required. Additionally, the model of the camera should be read in during staging and used to discern the value of the deadtime.

Relevant for any beamline using the Mako Driver with cameras other than the G-234 or any beamline using the Manta driver (when it is replicated in dodal).

stan-dot, 2024-12-02:
what are the relevant beamlines @DiamondJoseph ?

b01-1  and?

DiamondJoseph, 2024-12-03:
[image]
At least these beamlines: b01-1, 13-1, i22, p38, p45, training rigs (p46-p49), used as an OAV on other beamlines and used as a mock detector on the test beamlines.