DiamondLightSource/dodal #384: Port Slits to ophyd-async and expand fields

callumforrester, 2024-03-15:
Inspired by #377 

An ophyd-async version of `Slits` is required for the upcoming I22 experiment in June 2024 in order to populate the NeXus file with relevant metadata. It seems sensible to port the existing version into an ophyd-async class that remains compatible with hyperion rather than create two devices.

Additional fields need to be added to populate the NeXus file correctly (see below)

## Acceptance Criteria
* The following ophyd devices are ophyd-async: `S4SlitGaps`
* @DominicOram and the hyperion team are happy with their compatiblity
* The attributes required by the NeXus files are exposed: X_centre, Y_centre, X_size, Y_size

callumforrester, 2024-03-15:
Assumption to be confirmed is that I22 slits are the same as I03 slits, or at least that the IOCs can be modified to have the same naming conventions.

callumforrester, 2024-03-15:
Have just realised there's no overlap in fields needed by I22 and hyperion
- Hyperion
    - xgap
    - ygap 
- I22
    - X_centre, Y_centre, X_size, Y_size
 
so two options:

1. Have one device with all fields, extra config to customise output of `.read()`
2. Have two devices

Any thoughts @DominicOram or @olliesilvester?

DominicOram, 2024-03-15:
Ok, had a chat to controls, `xgap` is the same as `X_size` but PVs are annoyingly not standard. We will ask them to move to the I22 standard given that the other slits on i03 use this standard

DominicOram, 2024-03-15:
FYI https://jira.diamond.ac.uk/browse/I03-956

DominicOram, 2024-04-04:
I think we should be able to switch to these by the time they're written, controls are making the changes now

PR #431 (2024-04-25): Ophyd async slits
Fixes #384 

Port S4SlitGaps to ophyd-async. Add tests to prove it reads correctly. Also create a group object for bunches of slits. Add slits config for I22 and P38.

### Instructions to reviewer on how to test:
1. Run unit tests

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)