DiamondLightSource/dodal #389: IVU Compatible Insertion Device

callumforrester, 2024-03-18:
Follow-up to insertion device for I22 requirement in #377. I22 has an IVU insertion device, which is different to the CPMUs used by I03 and I04. Further consultation is needed to see if the PVs are sufficiently similar that the current [Undulator](https://github.com/DiamondLightSource/dodal/blob/c79e88295a066eb1630592b3fa031d6f033bafa8/src/dodal/devices/undulator.py#L16) device can be generalised (and ported to ophyd-async) or if it makes sense to create a new one.

@Tom-Willemsen I see you've done some work on I23, which has the same type as I22, did you ever look into making an ophyd device for the ID.

## Acceptance Criteria
* IVU device available
* Fields specified in #377 are exposed

Tom-Willemsen, 2024-03-18:
> @Tom-Willemsen I see you've done some work on I23, which has the same type as I22, did you ever look into making an ophyd device for the ID.

No, I don't think I ever looked at this - the work on i23 that I did was for a very narrow use-case (OAV pin-tip detection), I don't think any other devices were added.

callumforrester, 2024-03-25:
Have consulted with Controls, the I22 Undulator should behave similarly to the I03/I04 one from an EPICS perspective, so the existing ophyd device will hopefully work for both beamlines. In which case this ticket just becomes about porting it to ophyd-async.

DominicOram, 2024-03-26:
We would presumably want to rename from `Undulator` -> `InsertionDevice` to be more generic?

DominicOram, 2024-03-26:
Tested running the `ophyd` undulator against I22 and it works without changes, should be easy to convert

PR #511 (2024-05-24): Add NeXus-friendly soft signals to Undulator
Add soft signals to Undulator that provide static information needed for the NXInsertionDevice class in the NeXus standard: poles and length.

Fixes #389

### Instructions to reviewer on how to test:
1. Run unit tests
2. Run system tests against i22, i03, i04 and i04-1

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing dev [... cut]
closed 2024-05-24 by callumforrester