DiamondLightSource/dodal #397: Port DCM to ophyd-async

callumforrester, 2024-03-25:
Required for #377

I22 has a DCM, so can port the existing implementation and test it works for all beamlines. The linked issue shows the extra fields the device would need to expose.

## Acceptance Criteria
* DCM available
* Fields specified in #377 are exposed

DominicOram, 2024-03-26:
Some complications here to make a generic DCM:
* i03 only has one pitch/roll motor, i22 has two, one for each xtal. Does i22 use both, do you know @Tom-Willemsen? 
* i03 has a wavelength PV, but we only use the energy so this should be fine to remove
* All the temperatures are present for both devices but they have different PV names, suggest we ask controls to alias these to sensible names?

olliesilvester, 2024-03-26:
Given Dom's comment, I think we either need  a `CommonDCM` ophyd-async device which we inherit from for i22 and for i03, or give the DCM device initial parameters to create the different variations

Tom-Willemsen, 2024-03-26:
> i22 has two, one for each xtal. Does i22 use both, do you know @Tom-Willemsen?

Both `roll1` and `roll2` are configured in GDA and then referenced from beamline setup scripts / staff scripts - so yes.

olliesilvester, 2024-03-26:
I think if we anticipate loads of variations of DCM's then having one device with parameters if better. If there will only be a few variations then inheritance from a common DCM is neater

Tom-Willemsen, 2024-03-26:
My _instinct_ is to mirror what is being done on the controls side - if they are deploying effectively the same IOC with configurable prefix, then it makes sense to be one ophyd device, if they're deploying _different_ IOCs with separate DBs etc then they should be separate ophyd devices - otherwise we'll _keep_ running into cases where one IOC gets updated but not another.

(In some cases it might make sense to also try to persuade controls to move towards more common IOCs - but that's a discussion to have with controls people)

callumforrester, 2024-04-11:
@DominicOram Given `UndulatorDCM`, do that, the DCM and the Undular (#389) all need to be ported to ophyd-async in one PR?

DominicOram, 2024-04-11:
Annoyingly yes, I'm somewhat open to having separate `ophy`/`ophyd_async` devices if it's only for a very short time

callumforrester, 2024-04-11:
If we think the underlying IOC behavoir will be the same and its a case of just porting from ophyd to ophyd-async, I'm not hugely against one big PR, I still think that can fit into a sprint. If there's other work that needs doing on the devices before migration then I agree we should separate

closed 2024-05-07 by DominicOram