DiamondLightSource/dodal #399: Ophyd-async device for F-Switch

callumforrester, 2024-03-25:
Required for #377

We need to implement an F-Switch device exposing the fields required for I22 nexus files (see linked issue).

## Acceptance Criteria
* F-Switch available
* Fields specified in #377 are exposed

joeshannon, 2024-04-09:
The f-switch is more complicated than just four motors, it has additional screens relating to filters and beamsize.

It uses the "transfocator" support module.

There is an implementation in dodal already:
https://github.com/DiamondLightSource/dodal/blob/18277c8847f8d1c9a9d5d2e5cc1ee1fbb43a2c97/src/dodal/beamlines/i04.py#L141-L153

This could be converted to ophyd-async and moved to a common location as the device is currently in the i04 module.

Also need to work out what functionality is needed, e.g. the current iml allows for setting the filters but for now presumably they just want to read them.

The desired fields in #337 do not exist in their current GDA Nexus files.

callumforrester, 2024-04-09:
Also need to check the I04 transfocator PVs are the same as the I22 ones and if not get one set of PVs changed, as in #384

joeshannon, 2024-04-11:
@DominicOram It turns out that the i04 f-switch contains an additional component about the beam size which is not present on i22.

I'm not sure what the best approach to take is, split the current device to be composed of an optional beam size component or to make a new device?

DominicOram, 2024-04-11:
My first reaction would be to ask i22 if a beamsize calculator would be useful and port the logic over from i04. It looks pretty simple, take the required size, the beam energy and some other constants and put them in a quadratic equation:
```
record(ao, "BL04I-MO-FSWT-01:PARAM_A")
{
  field(VAL, "53.85")
}

record(ao, "BL04I-MO-FSWT-01:PARAM_B")
{
  field(VAL, "198")
}

record(ai, "BL04I-MO-FSWT-01:VERT_REQ") # this goes to edm
{
  field(VAL, "")
}

record(calc, "BL04I-MO-FSWT-01:LENS_PRED")
{
  field(VAL, "10")
  field(INPA, "BL04I-MO-FSWT-01:PARAM_A") 
  field(INPB, "BL04I-MO-FSWT-01:PARAM_B")
  field(INPC, "BL04I-MO-FSWT-01:VERT_REQ CP")
  field(INPD, "BL04I-MO-DCM-01:ENERGY.RBV CP") 
  field(CALC, "A/(C+B)*D**2") #check if same as (A/(C+B))*(D**2)
}

### Calculate the beamsize from nmber of lenses ###
record(calc, "BL04I-MO-FSWT-01:CALC_VER")
{
  field(VAL, "10")
  field(INPA, "BL04I-MO-DCM-01:ENERGY.RBV CP") 
  field(INPB, "BL04I-MO-FSWT-01:LENSES CP")
  field(INPC, "BL04I-MO-FSWT-01:PARAM_A")
  field(INPD, "BL04I-MO-FSWT-01:PARAM_B")
  field(CALC, "C/B*A**2-D") # check is same as ((C/B)*(A**2))-D
}
```
@JamesOHeaDLS - how easy would it be to put this somewhere common for i22 too?

closed 2024-04-29 by joeshannon