DiamondLightSource/dodal #404: Update adsim to ophyd_async

DiamondJoseph, 2024-03-28:
The AreaDetector simulator backed by the Epics AreaDetectorSimulator is split into three parts: `dodal/src/dodal/devices/adsim.py` contains a 5-axis MotorBundle; `dodal/src/dodal/devices/areadetector/adsim.py` contains a simulated detector; `dodal/src/dodal/adsim.py` is equivalent to a beamline. Having these maintained and using ophyd_async may enable fast testing of plan behaviour and would give an easily run up simulated beamline for local development.

## Acceptance Criteria
* Blueapi can be run up with the areadetector simulator devices exposed and connected with ophyd_async signals and an example grid_scan may be run with data being emitted

DominicOram, 2024-07-30:
The simulator is a generic EPICS thing, right? Would this make more sense inside `ophyd-async` @coretl?

DiamondJoseph, 2024-07-30:
https://github.com/bluesky/ophyd-async/tree/main/src/ophyd_async/epics/adsimdetector

There is an adsimdetector in ophyd-async

DominicOram, 2024-08-14:
As part of this make sure https://github.com/DiamondLightSource/dodal/pull/737#discussion_r1711422672 has coverage

PR #405 (2024-12-03): Update AdSim devices and configuration to use ophyd-async
Fixes #404 

### Instructions to reviewer on how to test:
1. Follow instructions for starting the AreaDetector simulator [included in the beamline module](https://github.com/DiamondLightSource/dodal/blob/adsim_async/src/dodal/beamlines/t01.py#L30) or from [the epics-containers docs](https://epics-containers.github.io/main/tutorials/launch_example.html)
2. Run the system tests

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [stan [... cut]
closed 2024-12-03 by DiamondJoseph