DiamondLightSource/dodal #418: Plane Mirror Device

stan-dot, 2024-04-04:
Plane mirror is used widely (possibly).
There are 2 instances at i20-1, the share those channels:
- Strip(Select) - enum field for the material element - ME2 has Silicon, Rhodium (10 mm or 17mm thickness),  ME1 has  Rhodium, Platium (-21 or -12 as positions) 
- Pitch (mrad)
- Y Position (enum in/out)
- Roll (mrad)
- Y (mm)
ME1 - does not have and rollw Y position, but has yaw and curvature.

And temperatures.

## Acceptance Criteria
TODO - ask how are those mirrors used, what is the most common space of movement.
At least 1 class of mirror has been created.

DominicOram, 2024-04-08:
This sounds similar to the existing [FocusingMirror](https://github.com/DiamondLightSource/dodal/blob/18277c8847f8d1c9a9d5d2e5cc1ee1fbb43a2c97/src/dodal/devices/focusing_mirror.py#L110C7-L110C21). Would be good to investigate if a common solution can be used for both

DominicOram, 2024-04-08:
See also https://github.com/DiamondLightSource/dodal/issues/398

stan-dot, 2024-04-17:
quite like #416 this is not the highest priority. Yet if other mirrors are faster and need ophyd device, that device could be uset here for the setup phase too.

@tomtrafford - how to frame best the answer to the 'what is the most common space of movement' question?