DiamondLightSource/dodal #434: Ensure we're getting debug logs from ophyd-async

DominicOram, 2024-04-11:
Following https://github.com/bluesky/ophyd-async/issues/195 we should make debug `ophyd-async` messages are correctly picked up and logged using the circular buffer as for `ophyd`

## Acceptance Criteria
* `ophyd-async` debug logs are logged the same way as `ophyd` ones

DominicOram, 2024-05-08:
* Currently `ophyd-async` has a debug logging framework but doesn't log anything
* This issue is to take the ophyd-async logging and set it's parent to our logging such that it's debug logs go through our CircularDebugHandler thing
* Inject some debug logging in some device
* Confirm this goes through into our logging handler + file
* Can test by mocking a handler and confirm that ophyd-async logs get into it when you call `.debug` on it

DominicOram, 2024-05-08:
We said small

olliesilvester, 2024-05-09:
I've made the changes for this, but I forgot to include the colorlog dependency in the ophyd_async change, so we need to wait for the next ophyd_async release before we can merge

PR #531 (2024-05-15): Add ophyd async logger to dodal logger
Fixes #434 

### Instructions to reviewer on how to test:
1. Confirm logs are correctly added and that tests pass

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
closed 2024-05-15 by olliesilvester