DiamondLightSource/dodal #437: Load dials env as a pre-launch task for tests

callumforrester, 2024-04-12:
As a developer, I want to run the unit tests via the vscode GUI without having to run the `load_dials_env` prelaunch task, as it is currently configured to do, so that I can develop against beamlines that are unrelated to dials without extra setup and develop in a non-dls_sw environment (e.g. my laptop or a devcontainer).

See below snippet from `launch.json`:

```json
           ...
            "purpose": [
                "debug-test"
            ],
            "console": "integratedTerminal",
            "preLaunchTask": "load_dials_env",
            "env": {
                // The default config in setup.cfg's "[tool:pytest]" adds coverage.
                // Cannot have coverage and debugging at the same time.
                // https://github.com/microsoft/vscode-python/issues/693
                "PYTEST_ADDOPTS": "--no-cov"
            },
           ...
```

@DominicOram @dperl-dls is the dials env still a requirement for dodal? If so then maybe we can make `load_dials_env` be a no-op if the environment is unavailable. 

## Acceptance Criteria
* The vscode test launch config no longer requires the fake dials env to start

callumforrester, 2024-04-12:
Sure, although thinking about it, should we make the vscode tests do exactly the same thing as `tox -e pytest`? I.e. `pytest -m 'not s03' {posargs}`

callumforrester, 2024-04-12:
Yeah that makes sense, I like the look of the hyperion setup. I'll make another issue for that, this one can remain just "make the dials env task ignore if unavailable"

stan-dot, 2024-12-02:
@callumforrester is this a still requested feature?