DiamondLightSource/dodal #438: Split tests into unit and system at the top level

callumforrester, 2024-04-12:
See comment in https://github.com/DiamondLightSource/dodal/issues/437#issuecomment-2051405124_
            
Mirror Hyperion's test structure: https://github.com/DiamondLightSource/hyperion/tree/main/tests

Have two top-level directories in `tests`: `system_tests` and `unit_tests` and organise the tests into those. Unit tests should be fast-running and have no external dependencies, so someone can easily clone the repo, setup a venv and run them with no extra steps. Systems tests should be everything else, including tests that require dials or s03.

## Acceptance Critera
- Tests split up similar to hyperion
- Running the unit tests, either through vscode or `tox -e pytest -- tests/unit_tests` is fast (ideally <30 seconds)
- The unit tests have no external dependencies

callumforrester, 2024-04-12:
Possibly could also include a test in CI that runs pytest against `unit_tests` without the `-m 'not s03'` arg, which would keep the unit tests independant

stan-dot, 2024-12-02:
since hyperion has been included into [mx-bluesky](https://github.com/DiamondLightSource/mx-bluesky/tree/main/tests).

also at the moment in Dodal we have a top level `system_tests` directory, maybe this is already done

callumforrester, 2024-12-02:
@DiamondJoseph could you make sure your adsim tests conform to that layout?

DiamondJoseph, 2024-12-03:
[image]
They do.

stan-dot, 2024-12-03:
> tox -e pytest -- tests/unit_tests 

just tried this at main @callumforrester  and I presume that you meant `-e tests` not `-e pytest`

it's definitely under 30 seconds, though test-cli at the moment runs connection to dodal.beamlines.i22 so that might need moving over to system tests

uncertain how to make sure that no external dependencies are there

callumforrester, 2024-12-03:
@stan-dot yes, the command has changed since dodal moved from the old python skeleton to the copier template, have edited

stan-dot, 2025-01-08:
isn't this already accomplished as of now?
https://github.com/DiamondLightSource/dodal

callumforrester, 2025-01-13:
There are still residual instances of `tests/../unit_tests/...`, do we want to get rid of those?

stan-dot, 2025-01-13:
`../unit_` search does not return anythin

callumforrester, 2025-01-15:
Are you sure about that? :)

```
> git clone git@github.com:DiamondLightSource/dodal.git
Cloning into 'dodal'...
remote: Enumerating objects: 208377, done.
remote: Counting objects: 100% (2867/2867), done.
remote: Compressing objects: 100% (1688/1688), done.
remote: Total 208377 (delta 2403), reused 1247 (delta 1175), pack-reused 205510 (from 3)
Receiving objects: 100% (208377/208377), 287.38 MiB | 8.85 MiB/s, done.
Resolving deltas: 100% (163816/163816), done.

> cd dodal/

> find . -name "unit_tests"
./tests/beamlines/unit_tests
./tests/devices/unit_tests
./tests/unit_tests
```

stan-dot, 2025-01-16:
you searched for `../unit` I failed to parse it as a general statement, I literally search for the string with two dots

stan-dot, 2025-02-04:
@DominicOram this would require a redo I think the PR got out of date