DiamondLightSource/dodal #452: Support selecting new visits per-request

DiamondJoseph, 2024-04-22:
#305 adds a DirectoryProvider implementation which is capable of mounting a single static visit per-beamline. All devices are configured to write to that visit directory, and a pre-processor may be used to update the dataCollection ("dcid") such that files are uniquely named (e.g. <ixx>-<dcid>-<deviceName>.h5, with dcid incrementing with each run. The visit may be updated by updating the dodal module and restarting the process. This is sufficient for current single-visit development and runs. 

When a more complete DirectoryService has been created, and authentication information can be passed as part of the request, it should be possible to select a visit per-request.

## Acceptance Criteria
* Multiple Runs can be started, each writing into a different visit directory on the same beamline without restarting the process.

callumforrester, 2025-01-20:
Relates to if not superseded by https://github.com/DiamondLightSource/blueapi/issues/552

oliwenmandiamond, 2026-09-08:
No longer relevant, done on blueapi side and `attach_data_session_metadata` has been removed https://github.com/DiamondLightSource/dodal/pull/2201

closed 2026-09-08 by oliwenmandiamond