DiamondLightSource/dodal #462: Rationalise Slits Implementations

callumforrester, 2024-04-24:
Dodal now has three attempts at a device to control 4-blade slits sets. One from #431, one from #315 and one originally imported from Hyperion: https://github.com/DiamondLightSource/dodal/blob/main/src/dodal/devices/s4_slit_gaps.py

As a developer I would like to rationalise them into a single device so I can reduce technical debt. See additional discussion in #384. Current work for the Hypersion slits and #431 requires underlying EPICS implementations to be rationalised first. From @DominicOram: this is being tracked internally at https://jira.diamond.ac.uk/browse/I03-956. Currently unsure if the #315 slits can also operate on the same of PVs, @dan-fernandes may be able to comment.

## Acceptance Criteria
* There is one class for 4-bladed slits
* It is an ophyd-async device
* It has tests
* All beamline configs that use slits now use it

DominicOram, 2024-05-05:
The new PVs are now on i03, we can switch over to using them

DominicOram, 2024-05-08:
To do the i03 + i22 merge:
* We need to change i03 over to use the i22 ones and see what breaks
* If there is specific logic in the i03 ones, move this over (but I don't think there is)
* Run i03 system tests to confirm
* Run hyperion unit tests
* Small job

DominicOram, 2024-06-26:
Blocked on the PVs not being as we expected. Working with controls on this

dan-fernandes, 2024-07-02:
Sorry, late to the thread.

> Currently unsure if the https://github.com/DiamondLightSource/dodal/pull/315 slits can also operate on the same of PVs, @dan-fernandes may be able to comment.

These slits implement a different Ophyd class for each set of slits that have different PV names, American vs British English, etc. (it was to just get stuff working, not very well thought out.)
Then my beamline-agnostic Bluesky plan had a factory method pattern to get the correct ophyd class from a config file.

There are other problems with my slit implementation (such as relying on a class I wrote that is essentially just a Motor, very rudimentary simulator, etc.) so I would be happy to edit my Bluesky plan to use whichever slit implementation you choose.

Happy to talk in person if this doesn't help!

olliesilvester, 2024-08-01:
mx-controls say they can make the PV naming consistent during next shutdown (CENTRE over CENTER)

stan-dot, 2024-08-05:
this reappered at i18

stan-dot, 2024-08-05:
also this is relevant
https://github.com/DiamondLightSource/dodal/issues/587

gilesknap, 2024-08-05:
I'd like to comment that in future I would expect all containerised motor IOCs with gap and centre slits to be using the template in the pmac module. So this is what we should be aiming for, or if not then we need to change it to what we want:
https://github.com/DiamondLightSource/pmac/blob/dls-master/pmacApp/Db/gap_and_centre_slits.template

rtuck99, 2025-01-06:
@olliesilvester Can you check if the required PVs have been added