DiamondLightSource/dodal #483: Make `device_factory` decorator to ease making and connecting ophyd-async Devices

coretl, 2024-04-29:
https://github.com/DiamondLightSource/dodal/issues/415#issuecomment-2079651830 summarizes a discussion on Device connection.
https://github.com/bluesky/ophyd-async/issues/265 will implement the ophyd-async supporting features of idempotent connect. The strategy we would like for dodal is:
- Each beamline module decorates its device factories with a `device_factory`
- This factory makes, names, caches and optionally connects a device
- In script use can pass `connect=True` to the device to connect in the foreground, requires a RunEngine to have been made
- In blueapi use can find all the eager devices and connect them at startup
- Can also connect them in a plan with the `ensure_connected` plan stub

This ticket is for:

## New ADR
Describing how Devices should be connected in dodal, from scripts and under blueapi

## `dodal.beamlines.beamline_utils`
```python
D = TypeVar(bound=Device)
class DeviceFactory(Protocol, Generic[D]):
    def __call__(self, connect: bool = False, timeout: float = DEFAULT_TIMEOUT) -> D:
        ...

_factory_made_devices: Dict[DeviceFactory, Device] = {}
_device_is_lazy: Dict[DeviceFactory, bool] = {}

def device_factory(lazy=False, set_name=True):
    def wrapper(f) -> DeviceFactory[D]:
        def factory(connect=False, timeout: float = DEFAULT_TIMEOUT):
            device = _factory_made_devices.get(f)
            if not device:
                device = f()
                if set_name:
                    device.set_name(f.__name__)
                _factory_made_devices[f] = device
            if connect: 
                call_in_bluesky_event_loop(device.connect(timeout=timeout))
            return device
        _device_is_lazy[factory] = lazy
        factory.__name__ = f.__name__
        return factory
    return wrapper

def get_device_factories() -> Dict[DeviceFactory, bool]:
    return _device_is_lazy.copy()
```
Need to add type hints to this so it knows the type of device it is making from the function it is decorating and check with @callumforrester that `get_device_factories` is what blueapi actually needs

## Acceptance Criteria
* ADR written so we know why this was done
* `device_factory` available to use in beamline modules
* `i22.py` updated to use it

stan-dot, 2024-06-06:
so the idea is that all `device()...` line in `i22.py` or other beamline gets this decorator, right?

coretl, 2024-06-07:
> so the idea is that all `device()...` line in `i22.py` or other beamline gets this decorator, right?

yes

stan-dot, 2024-06-18:
@coretl what is the role of keeping the lazy devices in a global dict? why is the key a complex object?

`def get_device_factories() -> Dict[DeviceFactory, bool]:
    return _device_is_lazy.copy()`
@callumforrester  how would blueapi use this? Maybe this should be stateful in some class, not a function in `beamline_utils`?

stan-dot, 2024-06-18:
```
ACTIVE_DEVICES: Dict[str, AnyDevice] = {}
BL = ""
DIRECTORY_PROVIDER: UpdatingDirectoryProvider | None = None

```
there is loads of global state in there! very surprising.

stan-dot, 2024-06-18:
[image]

classes graph

stan-dot, 2024-06-18:
`    already_existing_device: AnyDevice | None = ACTIVE_DEVICES.get(name)`
this line in device_instantiation makes most of the difference

coretl, 2024-06-18:
> @coretl what is the role of keeping the lazy devices in a global dict? why is the key a complex object?

As written in the original issue:

> - In blueapi use can find all the eager devices and connect them at startup
> - Can also connect them in a plan with the ensure_connected plan stub

So it looks like this:
- At startup connect in parallel `{device.connect() for device, lazy in get_device_factories().items() if not lazy}`
- When running a plan in parallel `{device.connect() for device in whatever_function_you _use_to_inject_devices_in_plan_args}`

PR #841 (2024-10-17): Isolated device factory
Fixes #483 

This is a re-implementation of #597 isolated to just the changes necessary to implement the `@device_factory()` decorator and have functions decorated with that decorator picked up by `make_all_devices`: respecting any configuration passed into the decorator.

NB: this is also required for [blueapi #647](https://github.com/DiamondLightSource/blueapi/issues/647)

### Instructions to reviewer on how to test:
1. System test the i22 beamline to ensure that devices are instantiated, connected and named appropriately.

### Checks for reviewer
- [ ] Would the PR t [... cut]
closed 2024-10-17 by DiamondJoseph