DiamondLightSource/dodal #487: Graylog: Improve message formatting

DominicOram, 2024-04-30:
Multiline messages do not format nicely on graylog, instead showing `/n` chars. See https://graylog.diamond.ac.uk/messages/graylog-daq-mx-hyperion_7/5018d5d6-06ef-11ef-9f8e-223dcd8d7ad4 we should try and work out if we can format them better

## Acceptance Criteria
* Investigate better formatting in graylog
* Fix if we can
* If this is a limitation of graylog document it somewhere as a known issue

DominicOram, 2024-05-07:
@dperl-dls thinks this may be fixed, check

dperl-dls, 2024-05-08:
oops, closed by accident sorry

DominicOram, 2024-05-08:
* We should make the log formatter take care of this and replace `/n` with newlines properly
* Timebox to 1 day, if we make no progress document on here why we had to stop and stop
*

dperl-dls, 2024-05-10:
It does look like with the most recent version (I think this is probably a python 3.11 thing) we do get e.g. tracebacks formatted correctly:

https://graylog.diamond.ac.uk/messages/graylog-daq-mx-hyperion_17/6069fe10-0ea0-11ef-a69c-fae0dce5bd43

Meanwhile the json is generally formatted compactly, this is because we insert it in strings ourselves. If we want this to be formatted the solution is to do it ourselves, with `json.dumps(..., indent=n)`:

https://graylog.diamond.ac.uk/messages/graylog-daq-mx-hyperion_17/741f64c5-0ea2-11ef-a69c-fae0dce5bd43

DominicOram, 2024-05-10:
Ah, if it's just the json then I think we should format it ourselves nicely for the logs

dperl-dls, 2024-05-10:
Well, we should probably add some solution for tracebacks in 3.10 if we are supporting it, but I don't think it's very urgent

closed 2024-05-16 by dperl-dls