DiamondLightSource/dodal #492: Broken tests: `set_readable_signals` does not accept `Motor`

DominicOram, 2024-05-01:
Tests are currently failing in `main` due to:
```
  File "/scratch/ffv81422/hyperion/dodal/src/dodal/beamlines/i22.py", line 119, in slits_6
    return numbered_slits(
           ^^^^^^^^^^^^^^^
  File "/scratch/ffv81422/hyperion/dodal/src/dodal/utils.py", line 101, in wrapper
    return func(*args, **kwds)
           ^^^^^^^^^^^^^^^^^^^
  File "/scratch/ffv81422/hyperion/dodal/src/dodal/beamlines/_device_helpers.py", line 27, in numbered_slits
    return device_instantiation(
           ^^^^^^^^^^^^^^^^^^^^^
  File "/scratch/ffv81422/hyperion/dodal/src/dodal/utils.py", line 101, in wrapper
    return func(*args, **kwds)
           ^^^^^^^^^^^^^^^^^^^
  File "/scratch/ffv81422/hyperion/dodal/src/dodal/beamlines/beamline_utils.py", line 102, in device_instantiation
    device_instance = device_factory(
                      ^^^^^^^^^^^^^^^
  File "/scratch/ffv81422/hyperion/dodal/src/dodal/devices/slits.py", line 18, in __init__
    self.set_readable_signals(
  File "/scratch/ffv81422/hyperion/dodal/.venv/lib/python3.11/site-packages/ophyd_async/core/standard_readable.py", line 68, in set_readable_signals
    self.add_readables(read, wrapper=HintedSignal)
  File "/scratch/ffv81422/hyperion/dodal/.venv/lib/python3.11/site-packages/ophyd_async/core/standard_readable.py", line 209, in add_readables
    obj = wrapper(readable)
          ^^^^^^^^^^^^^^^^^
  File "/scratch/ffv81422/hyperion/dodal/.venv/lib/python3.11/site-packages/ophyd_async/core/standard_readable.py", line 238, in __init__
    assert isinstance(signal, SignalR), f"Expected signal, got {signal}"
AssertionError: Expected signal, got <ophyd_async.epics.motion.motor.Motor object at 0x7f423024a090>
```

This is happening because `Slits` is passing a `Motor` into `set_readable_signals`, which only accepts a `Signal`. @coretl do we expect `Motor` to be able to be passed in? In which case is this an `ophyd-async` error?

## Acceptance Criteria
* Tests are passing on main

coretl, 2024-05-01:
After a long discussion in https://github.com/bluesky/ophyd-async/issues/192 we settled on https://github.com/bluesky/ophyd-async/issues/223, but implemented `set_readable_signals` on top of the new methods to ease adoption with a deprecation warning. However, it looks like the reimplemented `set_readable_signals` uses `HintedSignal` underneath which means we are now enforcing that it is a `Signal` as per the original type hints.

My preferred solution would be to switch from:
```python
        self.x_gap = Motor(prefix + "X:SIZE")
        self.y_gap = Motor(prefix + "Y:SIZE")
        self.x_centre = Motor(prefix + "X:CENTRE")
        self.y_centre = Motor(prefix + "Y:CENTRE")

        # Type ignore pending: https://github.com/bluesky/ophyd-async/issues/192
        self.set_readable_signals(
            read=[
                self.x_gap,
                self.y_gap,
                self.x_centre,
                self.y_centre,
            ]  # type: ignore
        )
```
to
```python
        with self.add_children_as_readables():
            self.x_gap = Motor(prefix + "X:SIZE")
            self.y_gap = Motor(prefix + "Y:SIZE")
            self.x_centre = Motor(prefix + "X:CENTRE")
            self.y_centre = Motor(prefix + "Y:CENTRE")
```
whenever you have non-signal readable children of a StandardReadable. Would that work for you?

DominicOram, 2024-05-01:
Yep, that works, thanks @coretl

PR #501 (2024-05-03): Update to ophyd-async v0.3a3
Fixes #492 

### Instructions to reviewer on how to test:
1. Do thing x
2. Confirm thing y happens

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
closed 2024-05-03 by DiamondJoseph