DiamondLightSource/dodal #494: Use tolerances in beamline parameters file for aperture scatterguard

DominicOram, 2024-05-01:
For the aperture scatterguard to be in position we currently use the motor tolerances, we should instead use those in the beamline parameters file see https://diamondlightsource.slack.com/archives/C048RD0KFFF/p1714579073626219

## Acceptance Criteria
* Tolerances from beamline parameters file are used in the aperture/scatterguard

dperl-dls, 2024-05-08:
Should there be signals for this or just attributes?

DominicOram, 2024-05-08:
* Signal for outside device, attribute otherwise. This is an attribute as will only be internal
* We already read the file, need to get the tolerances out as well as positions and use those
* This is just aperture scatterguard for now, ignore the other tolerances in this file
* Small

closed 2024-05-20 by dperl-dls