DiamondLightSource/dodal #502: Automatically keep I22 and P38 in sync

callumforrester, 2024-05-03:
As a developer I would like to keep P38, at least from an ophyd point of view, as a close representation of I22. Currently they are treated as different beamlines with some similar components.

## Possible Solution

Could we somehow define different versions of particular devices?

## Acceptance Criteria
* TBD

DiamondJoseph, 2024-05-03:
A possible solution is to use the same beamline module for both, with careful @skip_device usage and care for visit directories- p38 is only ever going to be a commissioning visit(?)

callumforrester, 2024-05-07:
I would suggest more than just `d11` and `saxs` in the same beamline module. I would suggest that we day `d11` _is_ `saxs`, or at least a testing representation of it. So the ophyd devices of P38 are a 1:1 representation of I22, even if we have to mock some devices.

stan-dot, 2024-07-15:
are the PV names the same? What is stopping us from using the file as-is?

should we have BEAMLINE_NAME derived from venv?

DiamondJoseph, 2024-07-15:
`BL = get_beamline_name("p38")`

Checks Environment Variables to see if BEAMLINE is exported (it will be on a beamline control machine).

Comparing `p38.py` and `i22.py`:

(best guess):
d11 ~ saxs: "-DI-DCAM-03:" vs. "-EA-PILAT-01:",
d12 ~ waxs: "-DI-DCAM-04:" vs. "-EA-PILAT-03:",
d3 ~ oav: "-DI-DCAM-01:" vs. "-DI-OAV-01:"

i0 ~ i0: "-EA-XBPM-01:" vs. "-EA-XBPM-02:"
linkam ~ linkam: "-EA-LINKM-02:" vs. "-EA-TEMPC-05",

panda1/2/3: match
panda4, it: no equivalent device on p38
all others: sim devices on p38

d11, d12 shouldn't be exposed as -EA-PILAT-01, as they aren't Pilatuses (? should we allow this? Hopefully all plan stubs are written for StandardDetectors?)
it should be added as a sim device or else repeating the values from i0 (?)

Definitely the Linkam and TetrAMM could be exposed on the same prefix

stan-dot, 2024-07-15:
then we have the manual option of just applying a ternary expression for the PVs /  or an if block 

or we could make some decorator or such.

we'd also need to make it crash for panda4 I guess if there is an attempt to use it in a plan???

callumforrester, 2024-07-16:
The specific case of a detector being a `PilatusDetector` if it's i22 and an `AravisDetector` if it's p38 is what's interesting here...

stan-dot, 2024-07-16:
ok why not just a ternary expression? 

not sure how different are the params, will explore this in a branch

callumforrester, 2024-07-17:
To add some concrete acceptance criteria:
- There should be one module, `i22.py`, `p38.py` should be deleted
- `dodal connect i22` should work without errors when the I22 PVs are available
- `dodal connect i22 --profile p38` or `BEAMLINE=p38 dodal connect i22` (syntax can differ) should work when the P38 PVs are available
- There should be no errors in either case, dodal should handle the differences between the beamlines

stan-dot, 2024-07-17:
I guessed some of those, but it's great to see this clearly. Thanks!

stan-dot, 2024-07-30:
`BEAMLINE=p38 dodal connect i22` will test this tomorrow, together with this
https://github.com/DiamondLightSource/i22-bluesky/issues/87

@DiamondJoseph please advise

stan-dot, 2025-01-17:
waiting for #921  to do a fresh PR on this

stan-dot, 2025-06-09:
@DiamondJoseph , @callumforrester is the spirit of [this PR ](https://github.com/DiamondLightSource/dodal/pull/862/files#diff-2fc5d9cd2f963cbcc90f0c0d8c9412f8ebd88d00e628719cc63a3b62cb24a3e4)  - that is, adding an early return depending on the 'is lab' condition per device or adding mock= is_lab in the `device_factory` call still correct?

stan-dot, 2025-06-10:
[image]

ah we just import `make_all_devices` from dodal into blueapi but those two different repos do not evolve together, we need to work with the abstractions - they might be leaking here.

As I looked into the import_module inside `blueapi` I see it uses with_config and there we'd use using source.module. however it'd still be just one module. 

The design constraints for this arguably i22 and p38 issue tie closely to the decisions made in the past by @callumforrester  who architected this. I might try to come up with something idiomatic using the Python module system, will add a PR but ultimately it does not look like the labs - beamline parallelism requirement was at the forefront of design two years ago. There is only so much that can be done now about this without a full project rewrite into a monorepo.

stan-dot, 2025-06-10:
point for closing the issue is that the relationship of _likeness_ is already between i22 and other beamlines and p38 is too part of that _likeness_ set. We could write a script to avoid comparing the ASTs manually, and do that in CI. That's more in line with the KISS principle. If we do some fancy overhead just to keep i22 and p38 in parallel automatically, we wind up with a complex thing that is unlike other setups across the set of all beamlines.

As the developer at i22 and p38 it's in your power to decide to close the issue @RJCD-Diamond imo

edit: dynamically changing the imported module is _black magic_ and imo we should stay away from it https://www.grumpymetalguy.com/programming/python_importlib/

stan-dot, 2025-06-10:
anyway, I'm washing my hands off this idea