DiamondLightSource/dodal #504: Remove or refactor axes_to_move

DiamondJoseph, 2024-05-03:
CF is fairly sure that axes_to_move was a Pydantic, APISchema or Mypy specific requirement (for type checking?): if it can be removed and the string value of an axis in a Spec converted to its underlying device, it would make typing a Scan with a Spec a little less verbose.

Alternatively, if a default factory was provided for the argument with {x: x for x in Spec.axes}, and the argument was allowed to be Optional.

DiamondJoseph, 2024-05-03:
This may have the additional complexity of being a requirement of blueapi loading the plan once the plan has been restored to dodal, rather than a dodal issue per-se.

There's also a future ticket once this has been done to adapt this to use StandardDetectors and continuous motion enabled with a motioncontroller.

stan-dot, 2024-05-03:
@DiamondJoseph could you link which files are relevant here?

DiamondJoseph, 2024-05-03:
https://github.com/DiamondLightSource/dls-bluesky-core

callumforrester, 2024-05-07:
Is there an issue to move these files into dodal in the first place?

DiamondJoseph, 2024-05-08:
There was a PR to move [them (and utils) into dodal with a comment that it came from the i22 post-mortem](https://github.com/DiamondLightSource/dodal/pull/304) but no issue. The utils got moved but the plans were dropped and restored to blueapi as short term for wrapped plans (once bluesky plans are annotated no longer required) and ScanSpec plans/FlyCount not mature enough implementation (this ticket is only really relevant once we can turn a ScanSpec into a FlyScan).

callumforrester, 2024-05-10:
Sorry for confusion, what's blocking this now?

stan-dot, 2024-07-22:
@coretl is it a good thing to fix now?

DiamondJoseph, 2024-07-29:
fwiw, I think the Pydantic deserialisation of plans may handle the following, but I haven't had a chance to test, where the Spec is constructed as Spec[str] on the client-side
```
def scan(
    detectors: List[Readable],
    spec: Spec[Movable],
    metadata: Optional[Mapping[str, Any]] = None,
) -> MsgGenerator:
```

DiamondJoseph, 2024-12-03:
Title is still fine but work was completed as part of moving the plan into dodal  #793

closed 2024-12-03 by DiamondJoseph