DiamondLightSource/dodal #507: Connection tests as CI cronjob

callumforrester, 2024-05-07:
As a developer I would like to be kept apprised of the ongoing accuracy of the system tests and hence the PV definitions on the beamline so that I can correct errors when PVs change.

Should consider how best to setup CI accounting for:
- Keeping PVs read-only so we don't affect the beamline
- Deciding whom to notify of specific failures
- One big CI job or one per beamline?

## Acceptance Criteria
* Local CI croniob that pulls dodal (all branches/PRs) and runs the connection tests (see #506) reporting any errors.

DominicOram, 2024-05-22:
I think we should really try and bump this up the priority and run it on PRs too. It would have caught https://github.com/DiamondLightSource/dodal/pull/480#issuecomment-2125156772

stan-dot, 2024-05-23:
> bump this up the priority

in jira or on the github project?

DominicOram, 2024-05-23:
Wherever the core team would like to prioritise it. For MX https://github.com/DiamondLightSource/dodal/issues/261 is probably a higher priority and is somewhat related so we can try and prioritise that.