DiamondLightSource/dodal #515: Only allow PRs to be merged if the PR branch is up to date with main

DiamondJoseph, 2024-05-09:
This is configurable in the github settings for the repository. It would prevent merged PRs from breaking main by being incompatible with changes that have gone in since the branch was taken.

## Acceptance Criteria
* The github UI prevents merging for a branch that does not extend from head of main.

DominicOram, 2024-05-09:
I agree with it, lets just do it

DiamondJoseph, 2024-05-09:
[image]

Here's the relevant option, can also require various checks to have passed before allowing a merge: should I enable this and enforce that testing on the latest version of python has passed?

DominicOram, 2024-05-09:
Yh, I think we can require that testing on all versions of python passed. Some of the other tests, like codecov, are potentially more flaky/controversial but the tests should all pass at least

DiamondJoseph, 2024-05-09:
Complete, PRs now require being up to date with main and also that the `test` action (which is the matrix of testing on all python versions) has passed.

closed 2024-05-09 by DiamondJoseph