DiamondLightSource/dodal #526: As discussed, we can make a separate issue to make sure these signals use their setters rather than the backends in order to set values. Also, there may be scope to have a regular `soft_signal_r` function in ophyd-`async`

callumforrester, 2024-05-13:
As discussed, we can make a separate issue to make sure these signals use their setters rather than the backends in order to set values. Also, there may be scope to have a regular `soft_signal_r` function in ophyd-`async`

_Originally posted by @olliesilvester in https://github.com/DiamondLightSource/dodal/pull/525#discussion_r1598549645_

callumforrester, 2024-05-13:
Soft signals now ship with setters that can be used in place of backends so there is no need to access `signal._backend`. Also mock signals (formerly known as sim signals) can be set with `set_mock_value(signal, value)`

Relm-Arrowny, 2024-05-14:
Can I pick this up, it broke CI and it is kind of annoying.

stan-dot, 2024-05-15:
I assume the totality of the changes here includes a deprecation of the `SimSignalBackend` from ophyd-async. We're using an OOP style in this repo therefore I would expect the class name to be an important keyword that if present, helps greatly to quickly make sense of the purpose and implications of the change.

shihab-dls, 2025-01-13:
Looking at this now, I don't believe any direct references to `_backend` remain; Callum, should we close this?

callumforrester, 2025-01-13:
@shihab-dls I can't see any either, feel free to do the honours

closed 2025-01-14 by shihab-dls