DiamondLightSource/dodal #529: Misleading slits device name

keithralphs, 2024-05-14:
https://github.com/DiamondLightSource/dodal/blob/050e9a4405e6a87e39843682df9963f6d8be3ced/src/dodal/beamlines/_device_helpers.py#L29

From @Tom-Willemsen: Should be called "slits" rather than "slit_gaps".

Tagging @callumforrester @joeshannon

Tom-Willemsen, 2024-05-14:
> From @Tom-Willemsen: Should be called "slits" rather than "slit_gaps".

Because the device has members for both gaps and centres - so I'd expect something like `slits_1` at the user level. Looks weird to talk to `s1_slit_gaps.x_centre` for example.

keithralphs, 2024-05-14:
Are you happy with the form s1_slits, s2_slits, etc.?

callumforrester, 2024-05-14:
I think `slits_1`, `slits_2` etc. is better as that mirrors the dodal factory function  names

Tom-Willemsen, 2024-05-14:
I'd prefer it to be the same as the dodal device name - i.e. `slits_1` I think.

Wasn't there discussion at some point about having `blueapi` automatically set the name of the device to the name in the dodal module anyway? Or perhaps I'm misremembering. If that's the case, then it would definitely be misleading to have different names depending on whether we're running in blueapi or not...

callumforrester, 2024-05-14:
@Tom-Willemsen there have been a few discussions around that, would be ideal if _dodal_ set the name to the function name so you don't have to supply it all, that would save a lot of boilerplate. Might come out in the wash as part of the tickets coming out of #415

PR #547 (2024-05-17): Fix slit naming
Fixes #529

### Instructions to reviewer on how to test:
Run unit tests which replicate the declaration as used in dodal/beamlines/i22.py but with a simulated ophyd device

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
closed 2024-05-17 by keithralphs