DiamondLightSource/dodal #539: Move some setup_oav utils from hyperion to dodal

DominicOram, 2024-05-15:
There are a number of setup utils for the oav, which would be useful to put into somewhere more common. Specifically `get_move_required_so_that_beam_is_at_pixel` from https://github.com/DiamondLightSource/hyperion/blob/862d2a2b033d1a650a68057466725dbe19889f6d/src/hyperion/device_setup_plans/setup_oav.py#L115 is very useful. 

## Acceptance Criteria
* `get_move_required_so_that_beam_is_at_pixel` and any associated dependencies and tests are moved into `dodal` `plans`
* Hyperion now points at these common plans

olliesilvester, 2024-05-30:
Is Dodal going to be used as a library for beamline-generic bluesky plans as well as a library for ophyd-async devices?

DominicOram, 2024-05-30:
Yes, it's fine to put some small plans in there if they are more to do with controlling specific devices than controlling the beamline as a whole

olliesilvester, 2024-05-30:
Makes sense, We could either have

olliesilvester, 2024-05-30:
As a convention for stuff like this, we could put useful plans for some device `a` into a file `a.plans`, or maybe `a.utils`? I don't really have a preference, but a convention would be good

PR #589 (2024-06-12): Move useful oav plans to dodal
Fixes #539

### Instructions to reviewer on how to test:
1. Confirm functionality
2. Is the structure sensible?

## **Points for discussion:**

- Where is the best place to put bluesky plans which are useful to any beamline using that device?
- Is the `WarningException` useful for anything other than Hyperion?

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing th [... cut]
closed 2024-06-12 by olliesilvester