DiamondLightSource/dodal #559: Prepare `ZocaloResults` to use multiple results sources

dperl-dls, 2024-05-20:
I03 will soon have both GPU-based and traditional xray-centring results available.  `ZocaloResults`  should fetch both of these from the queue and pick which one to use based on input from hyperion.

DominicOram, 2024-07-09:
Need to discuss with Nick on multiple messages vs single messages with both results

DominicOram, 2024-08-16:
@ndevenish says it will be multiple messages. For implementing this lets assume the format is:
```
{
    "results": [
        {
            "centre_of_mass": [
                2.207133058984911,
                1.4175240054869684,
                13.317215363511659,
            ],
            "max_voxel": [2, 1, 13],
            "max_count": 702.0,
            "n_voxels": 12,
            "total_count": 5832.0,
            "bounding_box": [[1, 0, 12], [4, 3, 15]],
        }
    ],
    "status": "success",
    "type": "3d",
    "source": "gpu",
}
```
It'll be close to that I suspect. Nick, feel free to correct us if you have a better idea of the format.

ndevenish, 2024-08-29:
You are listening to messages directly, right? Are you getting the messages back from the `xrc.{ispyb_beamline}` queue?

It might be in the message parameters, along with dcid/dcgid, is that okay?

olliesilvester, 2024-08-29:
As discussed with @DominicOram:

- We want a parameter to turn on GPU processing and a separate one for using the GPU results
- If `use_gpu_results` is on, then always use the first message we recieve from `ZocaloResults`. Otherwise, use the CPU results

olliesilvester, 2024-08-29:
As part of this issue, if we are using gpu results, we would like to compare data from GPU and CPU results and include it in a debug log, and warn if the results differ by some tolerance (TBC)

This should be done asynchronously so we aren't still waiting on the CPU results

PR #763 (2024-09-23): ZocaloResults: add parameter to use results from GPU
Fixes #559 

See related `mx_bluesky` PR: https://github.com/DiamondLightSource/mx-bluesky/pull/445

### Instructions to reviewer on how to test:
Check this PR adds and tests the following behavior:

If the GPU is toggled on, then:
- Wait for results twice
- If CPU arrives before GPU, then warn and continue
- if CPU results differ from GPU within a tolerance, log the two results and continue
- if GPU times out but CPU results arrive, warn and continue
- if GPU results arrive but CPU results timeout, error
- Always use CPU results after compa [... cut]