DiamondLightSource/dodal #570: Beamline prefix in *_01 beamlines are wrong

DominicOram, 2024-05-22:
For a beamlines with `_1` the PV prefixes are `BLXXJ` rather than `BLXXI` e.g. i20_1 has PVs of `BL20J`. However, when the `i20_1.py` module is used it incorrectly uses `BL04I`. This can be seen by running `dodal connect i20_1` after https://github.com/DiamondLightSource/dodal/pull/519 is merged

## Acceptance Criteria
* `dodal connect i20_1` correctly connects to devices

DominicOram, 2024-05-22:
FYI @stan-dot, you'll probably get round i20_1 before MX get round to any `_1` beamlines so you will need to fix this first

stan-dot, 2024-05-23:
`i20_1` beamline is on indefinite leave. The prefixes issue for it won't be a problem

DominicOram, 2024-09-24:
Duplicate of https://github.com/DiamondLightSource/dodal/issues/760 but that one is better written

closed 2024-09-24 by DominicOram