DiamondLightSource/dodal #593: Temorary I22 DCM

callumforrester, 2024-06-03:
I22 needs a DCM object urgently, the PVs arbitrarily differ from the existing DCM. Reconciliation in the future should be possible, I have made an issue: https://github.com/DiamondLightSource/dodal/issues/592

## Acceptance Criteria
* DCM class using I22 PV names
* Tests pass (including manual dodal CLI tests)

DominicOram, 2024-06-04:
This was caught on the original issue https://github.com/DiamondLightSource/dodal/issues/397#issuecomment-2020373884 is there something we could have done to make sure we didn't lose sight of it?

callumforrester, 2024-06-05:
Yes... I should have re-read it more carefully, sorry!
Beyond that I don't know, we should probably discuss...

The thing we didn't catch until now is that we actually _do_ need the wavelength PV (or to calculate it by other means), so that's a new problem to deal with, @DiamondJoseph

PR #598 (2024-06-13): 593 Temporary I22 DCM
Fixes #593 

### Instructions to reviewer on how to test:
1. Ensure tests pass
2. Ensure system tests pass

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests for the relevant beamline(s) been run via `dodal connect ${B [... cut]
closed 2024-06-13 by callumforrester