DiamondLightSource/dodal #610: `TetrammController` does not take an `arm` argument

callumforrester, 2024-06-11:
I get the following when trying to `count` a `TetrammDetector`:

```shell
TypeError("TetrammController.arm() missing 1 required positional argument: 'exposure'")>
```

Reference: https://github.com/DiamondLightSource/i22-bluesky/issues/35#issuecomment-2160438565

## Acceptance Criteria
* Tetramm can be `count`ed with a `RunEngine`
* Test to prevent regression

callumforrester, 2024-06-11:
The reason after discussion with @coretl is that the tetramm doesn't support software triggered scanning. When `trigger()` is called via a software scan it does not pass `exposure` to `controller.arm`. We will keep this status quo for now but should make the error message clearer.

joeshannon, 2024-06-11:
Note from discussion, also change type signature on `arm` to `exposure: float | None`.

PR #616 (2024-06-13): Improve error message if Tetramm arm called with no exposure time
Fixes #610

### Instructions to reviewer on how to test:
1. Run count plan with a tetramm
2. Confirm new exception is shown

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection tests [... cut]
commit acb949dc (2024-06-13): Improve error message if Tetramm arm called with no exposure time (#616)

The Tetramm only supports hardware triggering.

Fixes #610.
closed 2024-06-13 by joeshannon