DiamondLightSource/dodal #617: Document guidance for deciding at which layer a piece of logic should go

olliesilvester, 2024-06-12:
As discussed with @DominicOram , it would be nice to have a flowchart-style set of rules to decide whether a piece of logic should go into `EPICS`, `ophyd-async`, or `Bluesky`

Something like this:
- If the logic only affects one device, is simple, won't change often, and/or is useful for scientists to be able to run outside of bluesky, it should go in EPICS

- If it only affects one device, and the logic is complicated, and it may need occasional changes and enhancements, it should go in `ophyd-async`. For this statement, where we have composite `ophyd-async` devices, we can treat the top level device and everything beneath it as 'one device'

- If it affects multiple devices, and/or if the logic is tied to the experiment where changes are required often, it should go in `bluesky`. 

Tagging a few people who may have thoughts:
@callumforrester @abbiemery @coretl 

## Acceptance Criteria
* A set of rules to determine where logic should go is agreed upon and documented

callumforrester, 2024-06-13:
I like this, my only question is how strict to make it? Do we want a prescriptive flowchart, a set of approximate guidelines, or somewhere in-between?

DominicOram, 2024-06-13:
Approximate guidelines I think is fine. We're never going to be able to police something that's very strict

DominicOram, 2024-06-13:
I think where you say `ophyd-async` in the above do you actually mean `dodal` @olliesilvester?

olliesilvester, 2024-06-13:
Yeah, I really mean creating the devices using the ophyd-async library, rather than putting it into the repo

PR #1250 (2025-08-01): Add docs for where to put dodal logic
Fixes #617 

### Instructions to reviewer on how to test:
1. run `tox -e docs autobuild` and read the docs added
2. Furiously debate the wording

After writing this I noticed a very similar section in ophyd-async https://blueskyproject.io/ophyd-async/main/explanations/where-device-logic.html . This dodal page is more diamond specific, but I'm perhaps it's better to delete the overlap?

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](http [... cut]