DiamondLightSource/dodal #679: Move to pydantic v2

DominicOram, 2024-07-16:
We should make sure that all our dependencies etc. work on pydantic v2

## Acceptance Criteria
* We pin `pydantic>2`
* Dependencies are happy to install on `pip install`
* Unit tests work

DominicOram, 2024-07-16:
Currently blocked on:
* https://github.com/bluesky/scanspec/issues/122
* https://github.com/DiamondLightSource/python-zocalo/issues/247

dperl-dls, 2024-07-16:
Once we can, we should then update some of the way we use pydantic features, e.g.: https://github.com/DiamondLightSource/dodal/blob/fe3f48cc7e65e05ca50025d22a2161bef4eadba3/src/dodal/devices/detector/detector.py#L59

dperl-dls, 2024-08-13:
We are now blocked by scanspec requiring numpy>=2.0 and p4p requiring numpy<2.0

blueapi and nexgen also still pinned to pydantic<2

DominicOram, 2024-08-19:
Waiting for another release of p4p

DiamondJoseph, 2024-08-29:
I believe we're currently *not* blocked by p4p- only Numpy 2 support is. I'm going to make the required changes and check

dperl-dls, 2024-09-03:
@DiamondJoseph yes, but unfortunately the version of scanspec which supports pydantic 2 also requires numpy 2, so we are indirectly still blocked by that

DiamondJoseph, 2024-09-03:
Not since [ScanSpec 0.7.2](https://github.com/bluesky/scanspec/releases/tag/0.7.2)

PR #764 (2024-09-06): Update to Pydantic2 and ophyd_async 0.5.x
Fixes #679, paves route to #644 after p4p updated.

### Instructions to reviewer on how to test:
1. Run system tests to ensure devices still connect

### Checks for reviewer
- [ ] Would the PR title make sense to a scientist on a set of release notes
- [ ] If a new device has been added does it follow the [standards](https://github.com/DiamondLightSource/dodal/wiki/Device-Standards)
- [ ] If changing the API for a pre-existing device, ensure that any beamlines using this device have updated their Bluesky plans accordingly
- [ ] Have the connection test [... cut]
closed 2024-09-06 by DominicOram