DiamondLightSource/dodal #700: Write an ophyd-async device against the fastcs eiger

DominicOram, 2024-07-23:
We would like to move the `eiger` on to the new `fastcs` version as well as move our device support to `ophd_async`. We should do both of these at the same time. You can run a simulated `fastcs` eiger by running:

1. podman run --rm -it --net=host -v /dev/shm:/dev/shm -v /tmp:/tmp [ghcr.io/dls-controls/eiger-detector-runtime:latest](http://ghcr.io/dls-controls/eiger-detector-runtime:latest)
2. mkdir -p /tmp/opi && podman run --rm -it --net=host -v /tmp/opi:/epics/opi [ghcr.io/diamondlightsource/eiger-fastcs:latest](http://ghcr.io/diamondlightsource/eiger-fastcs:latest) ioc EIGER
3. module load phoebus && phoebus.sh -resource /tmp/opi/eiger.bob

Additionally, there is already one running on i04 at `BL04I-EA-EIGER-02:`, see https://diamondlightsource.slack.com/archives/C0176UKVBCK/p1721376023655459

i03 does not yet have the new eiger so we will need the old one around for a while still

## Acceptance Criteria
* There is an ophyd_async eiger device that replicates the same functionality as the current `ophyd` device
* We change `i04.py` to use this and `dodal connect i04` still passes

DominicOram, 2024-08-01:
More info on https://diamondlightsource.slack.com/archives/C0176UKVBCK/p1721376023655459

DominicOram, 2024-08-01:
* Try and mainatin plan level compatibility between this and the old one
* We think Large

DominicOram, 2024-08-15:
Gary successfully tested with
```
caput -S ODIN:FP:DatasetDataCompression BSLZ4
caput -S ODIN:FP:DatasetDataDatatype uint16
caput ODIN:FP:ProcessFramesPerBlock 1000
caput ODIN:FP:ProcessBlocksPerFile 1
caput -S ODIN:FP:ConfigHdfFilePath /tmp
caput -S ODIN:FP:ConfigHdfFilePrefix test
caput ODIN:FP:ConfigHdfFrames 3600
caput ODIN:FP:ConfigHdfWrite 1

caput -S EIGER:TriggerMode ints
caput -S EIGER:HeaderDetail all
caput EIGER:Nimages 3600
caput EIGER:FrameTime 0.01
caput EIGER:CountTime 0.0099
caput EIGER:Arm 1

caput EIGER:Trigger 1
```

DominicOram, 2024-08-15:
From discussion with Gary:
* The new IOCs bare no relation to AreaDetector
* This is hopefully cross-facility so put in `ophyd-async`
* Deployed on i04 is just the new eiger IOC, not the new odin one

DominicOram, 2024-09-02:
Initial MVP completed in https://github.com/bluesky/ophyd-async/pull/528. A number of spin off issues were made https://github.com/bluesky/ophyd-async/issues?q=is%3Aissue+is%3Aopen+eiger. I'm going to leave this issue here to track it from the downstream perspective but waiting on others to do https://github.com/bluesky/ophyd-async/issues/405 and https://github.com/bluesky/ophyd-async/issues/551 before continuing the work

olliesilvester, 2025-01-17:
> Odin deployment + tickit sim
> 
> ```
> podman run --rm -it -v /dev/shm:/dev/shm -v /tmp:/tmp --net=host ghcr.io/diamondlightsource/eiger-detector-runtime:1.16.0beta5
> ```

@shihab-dls and I were working through this and ran into permission issues. @shihab-dls managed to get around this by doing something like 

`podman run --rm -it -v /dev/shm:/dev/shm -v /scratch/some_directory/tmp:z --net=host ghcr.io/dls-controls/eiger-detector-runtime:latest`

Alternatively you can add `--security-opt label=disable` to the podman command

DominicOram, 2025-01-20:
> Alternatively you can add --security-opt label=disable to the podman command

The recommended workaround for this used to be disable `SELinux`, which I put a note about at https://github.com/bluesky/ophyd-async/tree/main/system_tests/epics/eiger. However, the dev portal no longer says how to do this this (since  https://gitlab.diamond.ac.uk/sscc-docs/developer-guide/-/commit/b696f01b33daf92ce0c9421572d4dd1b2dc0beb0). @cjreyn is this no longer recommended?

cjreyn, 2025-01-20:
Should be covered here: https://dev-portal.diamond.ac.uk/guide/containers/tutorials/podman/#mounting-volumes